AxisCommunications/acap-computer-vision-sdk-examples #15: Unable to run custom model in inference server

JoniSuominen, 2022-03-08:
Hi!

I recently tried to run a custom models with the inference client, but one gave me the following error:

`larod: Session 55: Could not run job: Could only read 196608 out of 786432 bytes from file descriptor`

I used the Python API, with the minimal-inference-server as a reference code base.
Inference on the model through Tensorflow Lite on my pc works fine.
Any tips on how to solve this?

Corallo, 2022-03-09:
Hi, could you ssh in the camera and run
journalctl -u larod

it might give us more informations about the issue

JoniSuominen, 2022-03-09:
@Corallo 
JournalCtl only gives the same message as I got before:
`Session 63: Could not run job: Could only read 196608 out of 786432 bytes from file descriptor`

JoniSuominen, 2022-03-10:
Im also curious about the chip id in environment files. If I want to run a TFLite model on the armv7hf CPU, which chip id do i choose? Previously I tried chip id 2 when using CPU models. Are these documented anywhere?

Corallo, 2022-03-10:
Regarding the chip, it is not properly documented yet, we are going to add a note in the readme, thanks again to report this issue.

In the meanwhile, yes, -c 2 makes the inference run on cpu. If you have an axis camera with an hardware accellerator you can use -c 4 to run on artpec7 tpu, or -c 12 to run on the artpec8 dlpu. 

Regarding the issue with your model, as you mentioned that the model you are using is "custom", please be aware that our deep learning API doesn't support custom layers, so it is not enough that the model runs on your local machine.

I'm in touch with the responsible team to see if we can provide you with more information to debug your specific issue, I'll give an update when I have news.

Corallo, 2022-03-10:
Another possiblity is that you are providing the wrong input to your model.
Maybe you supllied the input as uint8 while the model wanted a float32?
196608 [uint8] * 4 = 786432 [float32]

JoniSuominen, 2022-03-10:
That was actually it, I had forgotten to quantize the inputs of one model to int8.  Thanks for pointing out a dumb mistake!

closed 2022-03-10 by JoniSuominen