AxisCommunications/acap-computer-vision-sdk-examples #17: Start docker containers on camera bootup

JoniSuominen, 2022-03-17:
Hi!

I wonder if there is an easy way to start the docker containers automatically on the camera when booting up, so that no remote machine would be needed to start them manually? I'm looking for a similar feature to the Native API, but would like to use the C++/Python APIs instead.

marbali8, 2022-03-23:
Hello,
Thank you for reaching out! We are looking into it, and will let you know as soon as we have an answer.

marbali8, 2022-04-29:
Hello,

We have been investigating how to restart containers after reboot using the "restart" option on docker and docker-compose. There is some unexpected behaviour with some containers and we haven't been able to isolate the cause of it yet, but we will keep working on it and update with the progress in this issue.

marbali8, 2022-04-29:
Hello again,

There was a [fix](https://github.com/AxisCommunications/docker-acap/pull/57) made two days ago to the Docker ACAP where this works. The image is released under beta in DockerHub ([axisecp/docker-acap:latest_beta-armv7hf](https://hub.docker.com/layers/docker-acap/axisecp/docker-acap/latest_beta-armv7hf/images/sha256-d40286654722de8533a9e66e5a9d220c5344c56051b3717184160df56dbfb7a0?context=repo)). 
The way to indicate a container to restart after reboot is:

- For docker: adding `--restart always` to the run command.
- For docker-compose: adding `restart: always` to the services in docker-compose.yml.

Apart from "always", there are other values you can use, explained [here](https://docs.docker.com/config/containers/start-containers-automatically/#use-a-restart-policy).

Thank you for your contribution! Feel free to close this issue if it works for you.

JoniSuominen, 2022-05-03:
That sounds handy, thank you! I'll reopen if I need help again.

closed 2022-05-03 by JoniSuominen