AxisCommunications/acap-computer-vision-sdk-examples #23: Link to acap-dl-models Docker Hub image

mattias-kindborg-at-work, 2022-04-11:
### Describe the bug

In [`README`](https://github.com/AxisCommunications/acap-computer-vision-sdk-examples#docker-hub-images) we reference [acap-dl-models](https://hub.docker.com/repository/docker/axisecp/acap-dl-models). This link doesn't seem to point to a public image since it requires Docker Hub authentication. Can we change this into a link that works, or is this image no longer important?

Corallo, 2022-04-14:
I believe that acap-dl-models is not used anymore in any of the examples.
Also the other link should be updated to: axisecp/acap-runtime.

I can update those two lines, but I think it might be better to remove them, because that information is not helpful in the general readme, and it's easier to keep it up to date in the individual examples.

mikaelli-axis, 2022-05-04:
I agree, lets plan for removing it.