AxisCommunications/acap-computer-vision-sdk-examples #28: pose-estimator-with-flask example not working with Q1615 MM III (AXIS OS version 10.11.65)

axisvivek, 2022-05-11:
### Describe the bug

pose-estimator-with-flask example not working with Q1615 MM III (AXIS OS version 10.11.65)

### To reproduce

Follow the readme file from this project

### Screenshots

[image]

### Environment

- Axis device model: [Q1615 Mk III]
- Axis device firmware version: [10.11.65]
- Stack trace or logs: 
/usr/bin/acap_runtime: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.34' not found (required by /host/lib/liblarod.so.1)
- OS and version: [Ubuntu 20.04 LTS]

mikaelli-axis, 2022-05-17:
Thanks for the report, we're looking into what is causing this. If possible, try falling back to FW 10.10 to see if this resolves the issue for you.

axisvivek, 2022-05-30:
Thank you for suggesting the old firmware. It is working with AXIS OS 10.10.69.

closed 2022-07-07 by Corallo