AxisCommunications/acap-computer-vision-sdk-examples #29: web-server example is not working

axisvivek, 2022-05-13:
### Describe the bug

web-server example not working

### To reproduce
Follow the steps on Readme
### Screenshots

[image]

### Environment

- Axis device model: [e.g. Q1615 Mk III]
- Axis device firmware version: [10.11.65]
- Stack trace or logs: [e.g. Axis device system logs]
Monkey HTTP Server v1.5.6
Built : May 12 2022 12:11:47 (gcc 9.4.0)
Home  : http://monkey-project.com
[+] Process ID is 7
[+] Server socket listening on Port 2001
[+] 2 threads, 134217724 client connections per thread, total 268435448
[+] Transport layer by liana in http mode
[+] Linux Features: TCP_FASTOPEN SO_REUSEPORT
[2022/05/13 11:56:47] [  Error] Segmentation fault (11), code=1, addr=0x4
Aborted (core dumped)
- OS and version: [ Ubuntu 20.04 LTS]

mikaelli-axis, 2022-05-17:
Thanks for the report, what version of the sdk was used to build the crashing app?

axisvivek, 2022-05-27:
I don't have the detail about exact ACAP SDK. As per Readme.md in [acap-computer-vision-sdk-examples](https://github.com/AxisCommunications/acap-computer-vision-sdk-examples)/web-server/ ACAP4. If you check the docker file, it uses "FROM arm32v7/ubuntu:20.04 as runtime-image-armv7hf".

Any instruction to know this?

axisvivek, 2022-07-12:
Working with AXIS OS version 10.10.69 with command "docker -H tcp://$AXIS_TARGET_IP:$DOCKER_PORT run --rm -p 2001:2001 -it $APP_NAME"

closed 2022-07-12 by axisvivek