AxisCommunications/acap-computer-vision-sdk-examples #30: Is it possible to develop a native app in python?

diolegend, 2022-05-15:
Is it possible to develop a native app in python because the non-compatibility axis q1951-E with the docker acap?

mikaelli-axis, 2022-05-17:
It is possible though the building of the Python interpreter using the native-sdk is non-trivial and not something we have official support for. It has been discussed but we have nothing to share about this at the moment.

diolegend, 2022-05-17:
Will Axis Q1951-E support with acap computer vision sdk?

Corallo, 2022-06-10:
@diolegend There are no plans currently to add support for the computer vision sdk to artpec-7 cameras (like Q1951-E) that do not dispose of a DLPU.
The discussion that mikael mentioned is internal.

I am going to close this issue.

closed 2022-06-10 by Corallo