AxisCommunications/acap-computer-vision-sdk-examples #32: Is it possible to make the opencv.dnn module available in acap-computer-vision-sdk?

anigabriel, 2022-05-26:
Is it possible to make the opencv.dnn module available in acap-computer-vision-sdk for axis P3265-LVE Dome Camera?
"""
OpenCV modules:
    To be built:                 calib3d core features2d flann imgcodecs imgproc objdetect python3 video videoio
    Disabled:                    world
     Disabled by dependency:      gapi highgui java_bindings_generator js_bindings_generator ml objc_bindings_generator photo python_tests stitching
    Unavailable:                 dnn java python2 ts
     Applications:                apps
    Documentation:               NO
    Non-free algorithms:         NO
"""

Corallo, 2022-06-01:
Hello [anigabriel](https://github.com/anigabriel)
We don't plan at the moment to extend the computer vision sdk, but you can try to rebuild it with the flags you need
https://github.com/AxisCommunications/acap-computer-vision-sdk/blob/master/sdk/Dockerfile.aarch64#L11
Once you rebuild the computer vision sdk, you can use your custom sdk to build the example application:
https://github.com/AxisCommunications/acap-computer-vision-sdk-examples/blob/master/object-detector-python/Dockerfile#L8

Here is an example of how we added more libs to the native sdk:
https://github.com/AxisCommunications/acap-native-sdk-examples/issues/41

Be aware that building Opencv.dnn will allow you to use only the cpu to execute the inference, making it slow.
This is because the hardware accelerator doesn't have APIs for opencv.dnn.

Corallo, 2022-06-09:
I am converting this issue into a Q&A discussion