AxisCommunications/acap-computer-vision-sdk-examples #39: cannot install hello-world on P3265

egSat, 2022-06-09:
Hi there,
I'm trying to install this computer vision example, https://github.com/AxisCommunications/acap-computer-vision-sdk-examples/tree/master/hello-world
But I fail to install it, I get the error "no space on device". I attach an ssh session screenshot to the camera, showing the available space before and after trying to install the example.
Camera firmware is 10.10.73. Docker daemon 1.2.3
Attached screenshot before and after trying to install the example, and dockerlog from the camera

Thank you very much
[image]

Corallo, 2022-06-09:
Hello @egSat 

The space in the camera for applications is limited and containers are often hundreds of MB.
P3265 support microSD expansions, we recommend to use it and set the docker acap to use store the containers on the SD storage. 
You can do it on the web gui -> systems -> plain config -> search -> "docker" -> SD card support -> Yes.
You might also need to format the SD card to ext4 if it is not yet that format.
That can be done on system -> storage.

Corallo, 2022-06-09:
I am converting this issue into a Q&A discussion