AxisCommunications/acap-computer-vision-sdk-examples #44: How can I interpret this python exercise, please help.

juak0, 2022-06-13:
Entry – Sampling – Cycles –
Decisions – Strings

1) Quantity: 5 times
2) The data to work with are: file, name and surname, age, sex code, category code, category, salary, retirement.
a) File: 1,2,3… i+1 is automatically generated
b) Surname: all capital letters.
c) Name: first capitalized, rest lowercase
d) Age: values ​​between 21 and 60. Validate.
e) Sex code: validate F/M.
f) Job category code: validate A/B/C/D
g) Enter the number of overtime hours at 50%. – Enter the number of overtime hours at 100%. VALIDATE income, values ​​between 0 and 15
h) If the sex code is F, the sex is Female, otherwise it is Male.
i) If the job category code is A, the category is SALESMAN and the salary is 72000, if the job category code is B, the category is CASHIER and the salary is 75000; if the category code is C, the category is ADMINISTRATIVE and the salary is 82000, otherwise the category is MAESTRANZA and the salary is 52000.
j) Show surname and first name of the oldest person
k) Show last name and name of the person with the least amount of overtime at 50%
l) Count number of women, number of men.
m) Number of people in each category.
n) Accumulate salaries by category.
o) Calculate retirement: 11% of salary – Social work: 3% of salary
p) Calculate the value to be charged according to overtime
q) Show data and results

juak0, 2022-06-13:
I understand that the exercise has to be developed within a FOR loop, in addition to requesting data with input, and that I must use counters outside the loop.
I also have to use conditionals (IF, ELIF, ELSE).
but I don't know how to integrate it.
Please if someone can help me understand and structure the code to be able to do this exercise.

mikaelli-axis, 2022-06-14:
This question is out of scope for this repository.

closed 2022-06-14 by mikaelli-axis