AxisCommunications/acap-computer-vision-sdk-examples #50: How to load custom model in object-detector-python

egSat, 2022-07-07:
Hello,

I'm not able to load a custom model in a modified example of object-detector-python. From the Dockerfile.model I see that the current model is compiled to run on google coral so I transformed my model acordingly (based on yolov4-tiny). 
My custom model is fully quantized to tf-lite int8 and then postprocessed to be compatible with google coral by using edgetpu_compiler with success.
But after loading the docker image to the camera (P3265) it shows the following error:
ERROR in Inference: Failed to load model model.tflite (Could not load model: Could not build an interpreter of the model)
 <_InactiveRpcError of RPC that terminated with:
object-detector-python-object-detector-python-1  | 	status = StatusCode.CANCELLED
object-detector-python-object-detector-python-1  | 	details = ""
object-detector-python-object-detector-python-1  | 	debug_error_string = "{"created":"@1657181878.926597840","description":"Error received from peer ipv4:172.29.0.2:8501","file":"src/core/lib/surface/call.cc","file_line":1063,"grpc_message":"","grpc_status":1}"

### Environment

- Axis device model: P3265-LVE
- Axis device firmware version: 10.10.73
- SDK VERSION: 1.2

Thank you very much

anigabriel, 2022-07-07:
Hello 
I've got the same error  when I run AxisCommunications/acap-computer-vision-sdk-examples/object-detector-python/ example without any changes. 
"/usr/bin/acap_runtime: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /host/lib/liblarod.so.1)"

egSat, 2022-07-07:
I suspect version 10.11.76 is incompatible with sdk computer vision 1.2. In mi case the example object-detector-python without changes was working with firmware 10.10.73

Corallo, 2022-07-07:
Hello! yes there was an already reported bug with 10.11 or greater firmware, that has been fixed today.
There are some minor changes on the examples, mainly in the docker-compose file. Make sure to have them up to date with the master branch.

I'll close this issue, feel free to open another one if you encounter more problems.

closed 2022-07-07 by Corallo

JENNSHIUAN, 2022-08-02:
Hi, I have the same issue when I loaded my own tflite model. 
I have already replaced the path to my model in the [Dockerfile.model]
(https://github.com/AxisCommunications/acap-computer-vision-sdk-examples/blob/main/object-detector-python/Dockerfile.model)

There is the error I encountered:
[image]

error message:
 ERROR in Inference: Failed to load model train-608_best.tflite (Could not load model: Asynchronous connection has been closed)
 
<_InactiveRpcError of RPC that terminated with:
object-detector-python_1  |     status = StatusCode.UNAVAILABLE
object-detector-python_1  |     details = "failed to connect to all addresses"
object-detector-python_1  |     debug_error_string = "{"created":"@1659425957.027178280","description":"Failed to pick subchannel","file":"src/core/ext/filters/client_channel/client_channel.cc","file_line":3217,"referenced_errors":[{"created":"@1659425957.027173640","description":"failed to connect to all addresses","file":"src/core/lib/transport/error_utils.cc","file_line":165,"grpc_status":14}]}"

### **Environment**

- Axis device model: P3265-LVE
- Axis device firmware version: 10.11.76 (also tried 10.10.73 but didn't work)
- SDK VERSION: 1.2.1
- docker daemon: 1.2.3

Please help me, thanks in advance.