AxisCommunications/acap-computer-vision-sdk-examples #52: Communication with webpage to c++

awais40, 2022-07-12:
I was using net_http.h in SDK 3 for communicating from webpage to c++ via .cgi file, I found out it is no more supported in SDK 4. I have also checked there is a new way of communicating via Monkey server, which needs a major change in my application architecture. Is there is a way for using same .cgi way in SDK 4?

marbali8, 2022-07-13:
Hello!

Thank you for getting in touch with us. net_http.h was deprecated before SDK 2, so we recommend you to stop using it. Our web-server example shows how we communicate now between application and webpage using reverse proxy. This configuration is not only not limited to Monkey, but is compatible with other servers.

Corallo, 2022-07-15:
Moving this from issue into discussion.