AxisCommunications/acap-computer-vision-sdk-examples #72: object-detector-python fails on tag v1.2

sheh, 2022-08-10:
### Describe the bug

`object-detector-python` tag **v1.2** example fails with error:

```
acap_dl-models_1          | COPYRIGHT
acap_dl-models_1          | coco_labels.txt
acap_dl-models_1          | ssd_mobilenet_v2_coco_quant_postprocess.tflite
acap_dl-models_1          | ssd_mobilenet_v2_coco_quant_postprocess_edgetpu.tflite
object-detector-python_acap_dl-models_1 exited with code 0
inference-server_1        | /usr/bin/acap_runtime: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /host/lib/liblarod.so.1)
object-detector-python_inference-server_1 exited with code 1
object-detector-python_1  | object-detector-python connect to: inference-server:8501
object-detector-python_1  | <_InactiveRpcError of RPC that terminated with:
object-detector-python_1  | 	status = StatusCode.UNAVAILABLE
object-detector-python_1  | 	details = "DNS resolution failed for service: inference-server:8501"
object-detector-python_1  | 	debug_error_string = "{"created":"@1660143486.605863880","description":"Resolver transient failure","file":"src/core/ext/filters/client_channel/client_channel.cc","file_line":1324,"referenced_errors":[{"created":"@1660143486.605859080","description":"DNS resolution failed for service: inference-server:8501","file":"src/core/ext/filters/client_channel/resolver/dns/c_ares/dns_resolver_ares.cc","file_line":359,"grpc_status":14,"referenced_errors":[{"created":"@1660143486.605787880","description":"C-ares status is not ARES_SUCCESS qtype=A name=inference-server is_balancer=0: Domain name not found","file":"src/core/ext/filters/client_channel/resolver/dns/c_ares/grpc_ares_wrapper.cc","file_line":698}]}]}"
```

### To reproduce

Install and run object-detector-python example according to the [steps](https://github.com/AxisCommunications/acap-computer-vision-sdk-examples/blob/main/object-detector-python/README.md).

### Environment

- Axis device model: Q3538-LVE, P3265-LV 
- Axis device firmware version: 10.11.76, 10.11.65, 10.12.73
- OS and version: Ubuntu 20.04 LTS

### Additional context

Looks like `axisecp/acap-runtime:0.6-aarch64` is outdated.

Corallo, 2022-08-11:
Hello.
Yes, please don't use the tagged version 1.2.
There is a known bug that prevent it to work properly on fw >= 10.11.
The master branch doesn't have this issue

Corallo, 2022-08-11:
We just publish a [new tag 1.2.1](https://github.com/AxisCommunications/acap-computer-vision-sdk-examples/releases/tag/v1.2.1) with examples using the new sdk and acap runtime that fixes that problem.
Thanks for highlighting this issue.

closed 2022-08-11 by Corallo