AxisCommunications/acap-computer-vision-sdk-examples #77: Models does not detect objects CHIP_ID=12  SDK=1.2.1

sheh, 2022-08-16:
### Describe the bug

I have a detection model built using [tf models](https://github.com/tensorflow/models) (tf 1.15.5 to have per-tensor quantization). The model worked fine on 10.8.x firmware and SDK=1.2. After upgrade to 10.11.x and SDK=1.2.1 I see the following: model works as expected on cpu (CHIP_ID=2) and does not detect anything (low scores, random boxes) for artpec8 (CHIP_ID=12). Whereas object-detection-python example works fine in my environment for cpu and artpec8 both.
When I change CHIP_ID=12 to CHIP_ID=2 my application just starts working. 
There are no any crashes or errors in logs in case CHIP_ID=12 everything looks the same like in case CHIP_ID=2.

### Environment

- Axis device model: Q3538-LVE
- Axis device firmware version: 10.11.67
- OS and version: Ubuntu 20.04 LTS
- Version: SDK=1.2.1, axisecp/acap-runtime:aarch64-containerized

### Additional context

Could you provide advice what should I check to find the cause of the issue? Any test to get more information?

sheh, 2022-08-18:
The model is [spaghettinet](https://github.com/tensorflow/models/tree/master/research/object_detection#spaghettinet-for-edge-tpu).
Could it be an unsupported operation or something like that?

Corallo, 2022-08-18:
Hi @sheh 
We would have to investigate this issue, and it might take time.
Could you share with us a demo version of your model already trained?

sheh, 2022-08-19:
[Model file.](https://drive.google.com/file/d/12mprkYLlCykRBBPt_Lrl3Ek4E0muNOBM/view?usp=sharing)

I slightly modified `object-detection-python` example to print scores.
I built the images uploaded to the device and run:

`./docker-compose-acap --env-file ./config/env.$ARCH.artpec8 up`

get no detection:

```
object-detector-python_1  | Time for call to inference-server: 136 ms
object-detector-python_1  | 0 Objects found
object-detector-python_1  | Total time for image inference: 154 ms
object-detector-python_1  | Time for call to inference-server: 50 ms
object-detector-python_1  | 0 Objects found
object-detector-python_1  | Total time for image inference: 62 ms
object-detector-python_1  | Time for call to inference-server: 52 ms
object-detector-python_1  | 0 Objects found
object-detector-python_1  | Total time for image inference: 60 ms
object-detector-python_1  | Time for call to inference-server: 51 ms
object-detector-python_1  | 0 Objects found
```
then I just change `artpec8` to `cpu` and get detection:

`./docker-compose-acap --env-file ./config/env.$ARCH.cpu up`

```
object-detector-python_1  | Total time for image inference: 256 ms
object-detector-python_1  | Time for call to inference-server: 237 ms
object-detector-python_1  | 3 Objects found
object-detector-python_1  | dog 0.809
object-detector-python_1  | bicycle 0.793
object-detector-python_1  | truck 0.680
```
`docker-compose-acap:`
```
#!/bin/bash

set -x
set -e

TSL_ROOT=...

docker-compose --tlsverify --tlscacert=$TSL_ROOT/ca.pem \
        --tlscert=$TSL_ROOT/cert.pem \
        --tlskey=$TSL_ROOT/key.pem \
        -H tcp://${AXIS_HOST}:2376 $@
```
Also I added `IMAGE_PATH=dog416.png` to `docker-compose.yml` (`static-image.yml` looks outdated).

Corallo, 2022-08-19:
Could you try to upgrade your firmware to 10.12 and see if the problem persist?

sheh, 2022-08-19:
Did that the same result.

sheh, 2022-08-24:
Any thoughts about cause of the issues? It blocks me from release an app.

Corallo, 2022-08-25:
Hi @sheh 

I've been in touch with the team that will work on this issue to have an idea of when this issue will be handled.
Unfortunately, it is a complex problem, and we don't have the resources to check your specific model soon, it will probably be pushed until later than one month from now.

Probably the easiest solution for you is to stay with the firmware 10.8 until this gets fixed.

sheh, 2022-08-25:
Thanks! Got it.
Just a guess - could it be because of relu6 activation? There are not such activation in the model example but I have them in my model.

Corallo, 2022-08-25:
It might be, but not sure, also I don't see the relu6 on netron. 
We have a similar problem with the [pose estimation example](https://github.com/AxisCommunications/acap-computer-vision-sdk-examples/blob/main/pose-estimator-with-flask/Dockerfile.model) that uses Movenet.
We have a similar scenario, using the accelerator after FW 10.8 all the confidences printed are zeros.

Corallo, 2022-10-20:
HI @sheh 
We recently released a new firmware version for our camera (11.0).
It solved some similar issues with other models, where the inference was being executed, but the output was corrupt.
I suggest you to try your model with the latest firmware, to see if it solves also your issue.

Corallo, 2022-10-28:
Hello again @sheh 

I confirm that with the new firmware the problems seems to be fixed 
I can detect the dog, car and truck in the image running it on the accelerator.

Feel free to open a new issue or discussion if you have further questions

closed 2022-10-28 by Corallo