AxisCommunications/acap-computer-vision-sdk-examples #90: object-detector-python container build fail

bsrkhan, 2022-09-13:
I'm using the object-detector-python project as a base. I've made changes to the detector.py that uses some additional python packages shapely and pandas. I've edited the dockerfile to include this:
ARG ARCH=armv7hf
ARG SDK_VERSION=1.2
ARG REPO=axisecp
ARG RUNTIME_IMAGE=arm32v7/ubuntu:20.04

FROM $REPO/acap-computer-vision-sdk:$SDK_VERSION-$ARCH AS cv-sdk
FROM ${RUNTIME_IMAGE}
COPY --from=cv-sdk /axis/python /
COPY --from=cv-sdk /axis/python-numpy /
COPY --from=cv-sdk /axis/python-tfserving /
COPY --from=cv-sdk /axis/opencv /
COPY --from=cv-sdk /axis/openblas /

WORKDIR /app
RUN pip install requests
RUN pip install jsonpickle
RUN pip install shapely
RUN pip install pandas
COPY app/* /app/
CMD ["python3", "detector.py"]

When I build the container using this command: 
docker build . -t container --build-arg axisecp --build-arg armv7hf --build-arg arm32v7/ubuntu:20.04
The container fails to build, the output is attached

The container builds successfully with the other 2 packages I added, requests and jsonpickle. It looks like it fails to install numpy, but there is already a version of numpy copied from /axis/python-numpy. Is there anyway to successfully build the container with these packages?

Thanks 
[output.txt](https://github.com/AxisCommunications/acap-computer-vision-sdk-examples/files/9559448/output.txt)

Corallo, 2022-09-15:
Hello bsrkhan. 

We build a custom numpy to run using openblas on our camera.
As you see in the [dockerfile](https://github.com/AxisCommunications/acap-computer-vision-sdk-examples/blob/main/object-detector-python/Dockerfile#L15) we currently copy the built numpy manually in the example.
This way, pip is not properly aware that numpy is already installed, and when you install packages that have numpy as requirement you run into these kinds of problems.
We have in program to improve this system, but we can't provide an ETA yet.

TL;DR:
For the moment, you can try the following:
- identify the packages that need numpy as requirement, 
- find all their dependencies and install them with pip (making sure that none of these dependencies requires numpy).
- install the main package with --no-dependencies
It is dirty but hopefully works.

Good luck.

bsrkhan, 2022-09-15:
Okay that makes sense, can you tell me where I will be able to find out when the program is released?

I'm trying to install one library at a time, starting with pandas. Checking the dependencies I get this:
C:\Users\rkhan>pip show pandas
Name: pandas
Version: 1.4.4
Summary: Powerful data structures for data analysis, time series, and statistics
Home-page: https://pandas.pydata.org
Author: The Pandas Development Team
Author-email: pandas-dev@python.org
License: BSD-3-Clause
Location: c:\users\rkhan\appdata\local\programs\python\python310\lib\site-packages
Requires: numpy, python-dateutil, pytz
Required-by:

I updated the dockerfile to install python-dateutil and pytz, both of which install without issue.
ARG ARCH=armv7hf
ARG SDK_VERSION=1.2
ARG REPO=axisecp
ARG RUNTIME_IMAGE=arm32v7/ubuntu:20.04

FROM $REPO/acap-computer-vision-sdk:$SDK_VERSION-$ARCH AS cv-sdk
FROM ${RUNTIME_IMAGE}
COPY --from=cv-sdk /axis/python /
COPY --from=cv-sdk /axis/python-numpy /
COPY --from=cv-sdk /axis/python-tfserving /
COPY --from=cv-sdk /axis/opencv /
COPY --from=cv-sdk /axis/openblas /

WORKDIR /app
RUN python3 -m pip install --upgrade pip
RUN pip install requests
RUN pip install jsonpickle
RUN pip install python-dateutil
RUN pip install pytz
RUN pip install --no-dependencies pandas
COPY app/* /app/
CMD ["python3", "detector.py"]

I've added the --no-dependencies flag to pandas, but it is still trying to install dependencies. Is there something I'm missing?
[output1.txt](https://github.com/AxisCommunications/acap-computer-vision-sdk-examples/files/9578715/output1.txt)

Corallo, 2022-09-16:
It seems --no-dependencies doesn't exist anymore, use 
`pip install --no-deps pandas`  instead

bsrkhan, 2022-09-16:
I should have mentioned earlier I also tried using the --no-deps flag and --install-option="--no-deps", both still try to install numpy.

bsrkhan, 2022-09-20:
Is there any other way to get around this?

Corallo, 2022-09-21:
Hi 
Unfortunately, we didn't find a work around yet.
We saw that by adding 
`RUN touch /usr/lib/python3.8/site-packages/numpy-1.17.3.egg-info` to the Dockerfile
it seems you can "trick" pip to recognize the presence of numpy, so it is normally skipped if you try to install it and avoid the override, but this doesn't seem to work in the case of pandas, it seems it builds its dependencies regardless numpy is installed or not.

bsrkhan, 2022-09-21:
It doesn't seem to work for shapely either, I'll have to shelf this. When the program to improve this comes out, how can be notified?

Corallo, 2022-09-22:
We'll update this issue, when we'll be able to improve the pip packaging

bsrkhan, 2022-09-26:
Thanks for the help with my issue, I'll wait for the improved pip packaging.

Corallo, 2022-09-27:
@bsrkhan I just would like to mention that this issue have not been prioritized, so even if we plan to address this problem, unfortunately we can't give an ETA for when that will be done.

Corallo, 2023-01-10:
Hey @bsrkhan

Maybe you can take a look at [this issue](https://github.com/AxisCommunications/acap-computer-vision-sdk/issues/32#issuecomment-1375680692) @daniel-falk showed how you can install pandas. 

This will work however only on aarch64 devices.

daniel-falk, 2023-01-10:
Hi @bsrkhan,

Feel free to contact me on daniel.falk.1@fixedit.ai if you are interested in any further help. It should be possible to get `pandas` and `shapely` to work with the `armv7` cameras too by building it together with the SDK.

pataxis, 2024-11-01:
In ACAP 12.0, the Computer Vision solution has changed and building container applications for `arch64` is now done with public Python packages.
The `armv7hf` stopped at ACAP 4.15 mapping to AXIS OS 11.11.

closed 2024-11-01 by pataxis