AxisCommunications/acap-computer-vision-sdk-examples #107: Changing default detection model to run on Axis camera (Urgent)

BenJam29, 2022-10-17:
Hi everyone,

I was developing a computer vision pipeline on Axis camera model Q1656-LE Box. I installed Axis ACAP and Axis computer vision SDK using Docker and everything is functional when I use the default detection model which is SSD Mobilenet V2. (Screenshot below).
[image]

This detection model is configured in the following files as far as I learned. 

- app/object-detector-python/Dockerfile.model
```
`ARG ARCH=aarch64

FROM arm64v8/alpine as model-image-aarch64

FROM model-image-${ARCH}

# Get SSD Mobilenet V2
ADD https://github.com/google-coral/edgetpu/raw/master/test_data/ssd_mobilenet_v2_coco_quant_postprocess_edgetpu.tflite models/
ADD https://github.com/google-coral/edgetpu/raw/master/test_data/ssd_mobilenet_v2_coco_quant_postprocess.tflite models/
ADD https://github.com/google-coral/edgetpu/raw/master/test_data/coco_labels.txt models/
ADD https://github.com/google-coral/edgetpu/raw/master/test_data/COPYRIGHT models/

CMD /bin/ls /models`
```

- config/env.aarch64.artpec8
```
`MODEL_PATH=/models/ssd_mobilenet_v2_coco_quant_postprocess.tflite
INFERENCE_SERVER_IMAGE=axisecp/acap-runtime:0.6-aarch64
INFERENCE_SERVER_COMMAND=/usr/bin/acap_runtime -p 8501 -j 12 `
```

What I need to do is to replace this default detection model with other open-source model like EfficientDet Lite0 (from coral AI model zoo [here](https://coral.ai/models/object-detection/#trained-models)) or any other newly created model. I tried to change the two files above to accomodate these changes but the Axis device keeps loading the default model SSD Mobilenet V2 and returns an error. 

- app/object-detector-python/Dockerfile.model (Modified)
```
`ARG ARCH=aarch64

FROM arm64v8/alpine as model-image-aarch64

FROM model-image-${ARCH}

# Get EfficientDet 0 Model
ADD https://raw.githubusercontent.com/google-coral/test_data/master/efficientdet_lite0_320_ptq_edgetpu.tflite models/
ADD https://raw.githubusercontent.com/google-coral/test_data/master/efficientdet_lite0_320_ptq.tflite models/
ADD https://github.com/google-coral/edgetpu/raw/master/test_data/coco_labels.txt models/
ADD https://github.com/google-coral/edgetpu/raw/master/test_data/COPYRIGHT models/

CMD /bin/ls /models`
```

- config/env.aarch64.artpec8
```
`MODEL_PATH=/models/efficientdet_lite0_320_ptq_edgetpu.tflite
INFERENCE_SERVER_IMAGE=axisecp/acap-runtime:0.6-aarch64
INFERENCE_SERVER_COMMAND=/usr/bin/acap_runtime -p 8501 -j 12 `
```

Then when I run rebuild the object detector according to the instructions and run the detector, I received the following error:
[image]

Thank you for your help in advance.

garaujo23, 2022-10-17:
Try `docker system prune -f --volumes` to remove the model volume, should then have your new model

BenJam29, 2022-10-18:
Thanks a lot @garaujo23 for your comment. When I want to rebuild the object detector I use the following bash script. The command you mentioned is executed and old volumes are deteted. The code is attached below. 
```
#!/bin/bash

: ' Description:

object-detector-python
├── app
├── |- detector.py
├── |- dog416.png
├── docker-compose.yml
├── static-image.yml
├── Dockerfile
├── Dockerfile.model

detector.py - The inference client main program
dog416.png - Static image used with static-image.yml
docker-compose.yml - Docker compose file for streaming camera video example using larod inference service
static-image.yml - Docker compose file for static image debug example using larod inference service
Dockerfile - Build Docker image with inference client for camera
Dockerfile.model - Build Docker image with inference model
'

# cd to hello world app directory
SCRIPT_DIR="$( cd -- "$( dirname -- "${BASH_SOURCE[0]:-$0}"; )" &> /dev/null && pwd 2> /dev/null; )"
cd $SCRIPT_DIR/../apps/object-detector-python

# For arm64 (Q1656-LE the architecture is arm64)
export ARCH=aarch64
# Valid options for chip on aarch64 are 'artpec8' (hardware accelerator) or 'cpu'
export CHIP=artpec8

# Camera Setup
export AXIS_TARGET_IP=206.174.180.54
export DOCKER_PORT=7581

# Define APP name
export APP_NAME=acap4-object-detector-python
export MODEL_NAME=acap-dl-models

# Clean docker memory
printf "\n#################### Cleaning docker memory ... ####################\n"
sudo docker -H tcp://$AXIS_TARGET_IP:$DOCKER_PORT system prune -af

# Install qemu to allow build flask for a different architecture
printf "\n#################### Installing qemu to allow build flask ... ####################\n"
sudo docker run -it --rm --privileged multiarch/qemu-user-static --credential yes --persistent yes

# Build and upload inference client for camera
printf "\n#################### Building and uploading inference client for camera ... ####################\n"
sudo docker build . -t $APP_NAME --build-arg ARCH
sudo docker save $APP_NAME | docker -H tcp://$AXIS_TARGET_IP:$DOCKER_PORT load

# Build and upload inference models
printf "\n#################### Building and uploading inference models for camera ... ####################\n"
sudo docker build . -f Dockerfile.model -t $MODEL_NAME --build-arg ARCH
sudo docker save $MODEL_NAME | docker -H tcp://$AXIS_TARGET_IP:$DOCKER_PORT load
```
The output for rebuilding is: 

```$ ./BUILD_OBJECT_DETECTOR_PYTHON.sh 

#################### Cleaning docker memory ... ####################
[sudo] password for behnam:             
Deleted Containers:
e95cb52a1d0275bcce18b7a16e0553433cacd1ec1e64ecc7f0fe63846a59e7fd
3cc1d47d05013ca4ffa0f5aee2d417ead829180016f1814248cee49c68be2f65
bc2f39af66ba4706d67d13717ac2de4186813e9c65b11265875ad810826a3230

Deleted Networks:
[... 126 lines cut ...]
 ---> 979e8195a241
Step 8/14 : COPY --from=cv-sdk /axis/python-numpy /
 ---> Using cache
 ---> a7ff49c5a010
Step 9/14 : COPY --from=cv-sdk /axis/python-tfserving /
 ---> Using cache
 ---> dbb56f713381
Step 10/14 : COPY --from=cv-sdk /axis/opencv /
 ---> Using cache
 ---> f0447d4cacf4
Step 11/14 : COPY --from=cv-sdk /axis/openblas /
 ---> Using cache
 ---> b57f43d55de9
Step 12/14 : WORKDIR /app
 ---> Using cache
 ---> 9abcf84e5341
Step 13/14 : COPY app/* /app/
 ---> Using cache
 ---> a686434caf70
Step 14/14 : CMD ["python3", "stage-1.py"]
 ---> Using cache
 ---> d5e0a31dd793
Successfully built d5e0a31dd793
Successfully tagged acap4-object-detector-python:latest
```

Corallo, 2022-10-19:
Hi @BenJam29 
The answer of @garaujo23 is correct, you need to make sure that the old volumes in your camera are removed.
First of all, make sure to run the command 
`docker-compose --tlsverify --host tcp://$DEVICE_IP:$DOCKER_PORT --env-file ./config/env.$ARCH.$CHIP down --volumes`
when you terminate your execution, to leave the camera in a "clean state"
Try to run it now, and see if it already solves your issue,
Otherwise run
`docker -H  tcp://$DEVICE_IP:$DOCKER_PORT volumes ls` to check if there are volumes left
and 
`docker volume prune -f` to do a forced cleaning

BenJam29, 2022-10-19:
Just tried to look deeper into the issue. I persistently receive the "<_InactiveRpcError of RPC that terminated with:status = StatusCode.DEADLINE_EXCEEDED" by using various models. Image attached below.  
[image]

I took a look at gRPC to have a better understanding of the problem but I couldn't resolve the issue yet. 

Any help would be appreciated. 
Best,

Corallo, 2022-10-20:
HI @BenJam29 

I am going to move this issue into the discussion tab, as this is not an issue nor an enchantment.