AxisCommunications/acap-computer-vision-sdk-examples #114: Error with docker-compose

ahosnyyy, 2022-10-26:
Running the command: docker-compose --tlsverify --host tcp://$DEVICE_IP:$DOCKER_PORT --env-file ./config/env.$ARCH.$CHIP up 

Gives me an error: 

[+] Running 0/1
 - inference-server Error                                                                                         15.8s
Error response from daemon: Get "https://registry-1.docker.io/v2/": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)

Camera: AXIS Q1615 MK III

Corallo, 2022-10-27:
Hello @ahosnyyy 

Which example are you trying to execute?
Seems that your camera can't reach the docker registry, is it possible that you have some network issue that prevents the camera to go outside?
If that's the case, you can try to pull the inference server by yourself and load it on the camera the same way you did it with your app.

ahosnyyy, 2022-11-01:
> Hello @ahosnyyy
>
> Which example are you trying to execute? Seems that your camera can't reach the docker registry, is it possible that you have some network issue that prevents the camera to go [...]

Thank you, I pulled the inference server by myself and loaded it to the camera.

closed 2022-11-01 by ahosnyyy