AxisCommunications/acap-computer-vision-sdk-examples #129: "RUN  apt-get" in Docker file gives error: "unknown instruction APT-GET"

amararjun, 2023-01-18:
> **Please do not disclose security vulnerabilities as issues. See our [security policy](../../SECURITY.md) for responsible disclosures.**

### Describe the bug
While trying "opencv-image-capture-cpp" and " parameter-api" the docker build gives an error

docker build --tag $APP_NAME --build-arg ARCH .
```
Sending build context to Docker daemon  17.92kB
Error response from daemon: dockerfile parse error line 15: unknown instruction: APT-GET
```
### To reproduce

Please provide as much context as possible and describe the *reproduction steps* that someone else can follow to recreate the issue on their own.

A team member will try to reproduce the issue with your provided steps. If there are no reproduction steps or no obvious way to reproduce the issue, the team will ask you for those steps. Bugs without steps will not be addressed until they can be reproduced.

Steps to reproduce the behavior:

1. Set up '...'
2. Do this '...'
3. See error

### Screenshots

If applicable, add screenshots to help explain your problem.

### Environment

- Axis device model: [e.g. Q1615 Mk III]
- Axis device firmware version: [e.g. 10.5]
- Stack trace or logs: [e.g. Axis device system logs]
- OS and version: [e.g. macOS v12.2.1, Ubuntu 20.04 LTS, Windows build 21996.1]
- Version: [version of the application, SDK, runtime environment, package manager, interpreter, compiler, depending on what seems relevant]

### Additional context

Add any other context about the problem here.

Corallo, 2023-01-18:
You are probably using an old version of docker that doesn't have buildkit enabled by default.
You can try to add `DOCKER_BUILDKIT=1` in front of your docker build command

Corallo, 2023-01-18:
Also, to avoid further problems, update your firmware to the latest version. The 1.5 version of the sdk have a breaking change. It won't work with fw lower than 11.0

amararjun, 2023-01-19:
@Corallo Thank you for ,as always, the prompt reply. It solved the two errors I encountered.

closed 2023-01-19 by amararjun