AxisCommunications/acap-computer-vision-sdk-examples #141: Deploying YOLOv8 to ARTPEC-8 cameras

philippe-heitzmann, 2023-04-10:
### Can ARTPEC-8 cameras run YOLOv8?

Could the Axis team please advise if ARTPEC-8 DLPU chips are able to support running int8 per-tensor quantized YOLOv8m? For reference our team is seeing model loading errors when attempting to load this model to the inference server leveraging the [object_detector_python](https://github.com/AxisCommunications/acap-computer-vision-sdk-examples/tree/main/object-detector-python) script, with these errors pointing to the model containing > max 16 graph partitions and certain operations not supported on ARTPEC-8 cameras (see docs for full overview of [YOLOv8](https://github.com/ultralytics/ultralytics/issues/189) architecture / layers), which we were curious to learn more about why this would break. Any pointers on this would be much appreciated if possible, thank you. 

```
inference-server_1        | ERROR in Inference: Failed to load model yolov8m_int8.tflite (Could not load model: Model contains too many graph partitions (137 > 16) and 68 of the graph partitions can't be run on the device. Consider redesigning the model to better utilize the device. (Is it fully integer quantized? Is it using non-supported operations?))
```

### To reproduce

1. Export YOLOv8m to int8 per-tensor quantized .tflite weights using exporter.py script made available [here](https://github.com/ultralytics/ultralytics/blob/main/ultralytics/yolo/engine/exporter.py)
Quantization parameters:
- int8
- per-tensor
3. Deploy exported .tflite weights to camera 
4. Run docker-compose with below command leveraging [object_detector_python](https://github.com/AxisCommunications/acap-computer-vision-sdk-examples/tree/main/object-detector-python) scripts
```
docker compose --env-file ./config/env.$ARCH.$CHIP up
# env file:
ARCH=aarch64
CHIP=artpec8
```
5. Observe above failed model loading errors 

### Environment

- Axis device model: AXIS P3267-LVE Dome Camera
- Axis device firmware version: 11.3.70
- Stack trace or logs: Console output when running 
- OS and version: [e.g. macOS v13.2.1]
- Client and server application scripts: [object_detector_python](https://github.com/AxisCommunications/acap-computer-vision-sdk-examples/tree/main/object-detector-python)

Corallo, 2023-04-11:
Hello @philippe-heitzmann 

Can you show the command that you used to run the export.py script and produce a tflite quantized per tensor?
Could you maybe also share the tflite that you get from the conversion?
That error is introduced to prevent users to use a model that has many nodes (>16) that fall back to the CPU in the middle of the graph. Since falling back to the CPU has a big time overhead, doing it more than ~16 times would make it so inefficient to make it not worth to use the accelerator, but rather run it directly on the CPU.
In your case, after the quantization, your graph have many of these nodes. It is not clear from the architecture description what is the problem specifically, you would need to look at the tflite and see if there are some floating point layers.

Corallo, 2023-04-13:
Hello @philippe-heitzmann 

Running` journalctl -u larod `after trying to load the model (and failing) gives you more info about the problem with the model.
Specifically, running it after loading your model you'll see 
`Apr 13 11:17:43 axis-b8a44f277efe sh[1156]: ERROR: hybrid data type is not supported in conv2d.`
This means that the conv2d are quantized only in the kernel parameters, but the convolution excepts input and produce outputs as float. This is not supported.
Besides, looking at your model with [netron](https://netron.app/) you can see that other layers are not quantized, like the Add and Mul nodes, this will make the execution fall back to the cpu after each convolution, and that's why you see that error saying that the graph is divided in 60+ pieces.
[image]

Corallo, 2023-04-14:
@philippe-heitzmann 
I took another look at the onnx2tf tool, but I tried it on yolov5, maybe this will still apply to you:
Running
` onnx2tf -i ./yolov5s.onnx -oiqt -qt per-tensor -ioqd uint8`
produce several tflite models 
```
yolov5s_relu_dynamic_range_quant.tflite
yolov5s_relu_float16.tflite
yolov5s_relu_float32.tflite
yolov5s_relu_full_integer_quant.tflite
yolov5s_relu_full_integer_quant_with_int16_act.tflite
yolov5s_relu_integer_quant.tflite
yolov5s_relu_integer_quant_with_int16_act.tflite
```
The correct quantization is `yolov5s_relu_full_integer_quant.tflite`
Maybe you can try the same, exporting the model from yolov8 to onnx and quantizing by yourself with onnx2tf.

Let us know how that works, we would be happy to hear that you succeeded.