AxisCommunications/acap-computer-vision-sdk-examples #153: acap_dl-models_1    | exec /bin/sh: exec format error pose-estimator_1    | exec /usr/bin/python3: exec format error

DucTaiVu, 2023-07-14:
> **Please do not disclose security vulnerabilities as issues. See our [security policy](../../SECURITY.md) for responsible disclosures.**

### Describe the bug

Explain the behavior you would expect and the actual behavior.

### To reproduce

Please provide as much context as possible and describe the *reproduction steps* that someone else can follow to recreate the issue on their own.

A team member will try to reproduce the issue with your provided steps. If there are no reproduction steps or no obvious way to reproduce the issue, the team will ask you for those steps. Bugs without steps will not be addressed until they can be reproduced.

Steps to reproduce the behavior:

1. Set up '...'
2. Do this '...'
3. See error

### Screenshots

If applicable, add screenshots to help explain your problem.

### Environment

- Axis device model: [e.g. Q1615 Mk III]
- Axis device firmware version: [e.g. 10.5]
- Stack trace or logs: [e.g. Axis device system logs]
- OS and version: [e.g. macOS v12.2.1, Ubuntu 20.04 LTS, Windows build 21996.1]
- Version: [version of the application, SDK, runtime environment, package manager, interpreter, compiler, depending on what seems relevant]

### Additional context

Add any other context about the problem here.

danielmyh, 2023-07-14:
Hi DucTaiVu,

The error could be due to a mismatch between architectures, and if so you can use the --platform flag [see here](https://docs.docker.com/build/building/multi-platform/) to do a multi-arch build to target the right architecture. If you think it is something else it would be great if you could add some more details about the error and how it occurred with steps, and we can help you better.

DucTaiVu, 2023-07-18:
Hi Danielmyh,
I'm fixed this bug, cause i ran the build app ~$ docker build --tag $APP_NAME --build-arg ARCH . then replaced it with admin rights ~$ sudo docker build --tag $APP_NAME --build-arg ARCH .
This gave me the error.
I fixed it as follows:
~$ sudo chmod 666 /home/user/.docker/buildx/activity/default
Then: 
~$ docker build --tag $APP_NAME --build-arg ARCH .
Thanks,

closed 2023-07-18 by Corallo