AxisCommunications/acap-computer-vision-sdk-examples #164: Issue while uploading docker images with no SD card support

ThenoobMario, 2023-10-06:
Hello, 

I am trying to upload my docker images to P3265-LVE (with 8 GB flash) camera running firmware 11.5.64. My docker images have a total size of < 700 MB. I compiled them in an eap file with ACAP docker-compose installed prior to installing my current app. However, I can still not upload them on the camera with no SD card.

Is there an issue with the way docker-compose handles storage space?

Corallo, 2023-10-06:
Hello @ThenoobMario 

It is not clear what you are trying to do.
Did you convert your docker image into an EAP file?  How? (and why? You could run the docker container directly on the device.)
How are you trying to install the eap file? Did you follow any example?

ThenoobMario, 2023-10-06:
Hello @Corallo,

Thank you for responding. There are two things I am trying to do. 

1. I am trying to load the docker images on the camera without an SD Card. 
2. I created an eap file by following the container-example of acap native SDK (Given in the native SDK example repo).

The main issue I have right now is, the docker files don't load onto the axis camera. I tried the `docker load` method proposed in the readme as well. It says the device doesn't have enough storage. Considering my Axis Camera has a flash of 8GB, I assume it shouldn't have been the case. Am I missing something here?

Corallo, 2023-10-06:
If you want to use a container example, the SD card is a requirement. The device doesn't have enough flash to host a container application.
The 8Gb that you see is the full flash memory, most of it is used by the Axis OS.
The docker container engine without an SD card have usually space for no more than 500-600 Mb on a clean system.

Packaging it in an eap file should not fix the issue, as you are just zipping the whole container in an archive-like.

Your options are:
- Install an SD card.
- Build your application using the native sdk instead of the computer vision sdk (see [native-sdk-examples](https://github.com/AxisCommunications/acap-native-sdk-examples/tree/main)).
- Try to shrink your container with something like [docker slim](https://github.com/slimtoolkit/slim)

Corallo, 2023-10-06:
Converting issue in a discussion as it is not a bug.