AxisCommunications/acap-computer-vision-sdk-examples #178: Object detection sample mounting error: Are you trying to mount a directory onto a file (or vice-versa)?

sreid-brock, 2024-04-09:
#### Minimum Debug Steps
Before opening the issue, try to load the model on the device using this command:

Please provide us this log together with your issue.

brock@ubuntu:~/Desktop/Axis/acap-computer-vision-sdk-examples/object-detector-python$ larod-client -g model_path -c cpu-tflite
larod-client: command not found
brock@ubuntu:~/Desktop/Axis/acap-computer-vision-sdk-examples/object-detector-python$ journalctl -u larod
-- No entries --

I am attempting to go through the steps for the object-detector-python sample, but the application fails to run. I was able to run the hello-world sample, but for object-detector-python I get an error on the docker compose command:

Error response from daemon: failed to create task for container: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error mounting "/usr/lib/libvdostream.so.1" to rootfs at "/usr/lib/libvdostream.so.1": mount /usr/lib/libvdostream.so.1:/usr/lib/libvdostream.so.1 (via /proc/self/fd/6), flags: 0x5000: not a directory: unknown: Are you trying to mount a directory onto a file (or vice-versa)? Check if the specified host path exists and is the expected type

### To reproduce

cd ~/Desktop/Axis/acap-computer-vision-sdk-examples/object-detector-python
export ARCH=armv7hf
export CHIP=tpu
export APP_NAME=acap4-object-detector-python
export MODEL_NAME=acap-dl-models
docker run --rm --privileged multiarch/qemu-user-static --credential yes --persistent yes
docker build --tag $APP_NAME --platform linux/arm/v7 --build-arg ARCH .
docker build --file Dockerfile.model --tag $MODEL_NAME --platform linux/arm/v7 --build-arg ARCH .
docker save $APP_NAME | docker load
docker save $MODEL_NAME | docker load
docker compose --env-file ./config/env.$ARCH.$CHIP up

See error:

WARN[0000] /home/brock/Desktop/Axis/acap-computer-vision-sdk-examples/object-detector-python/docker-compose.yml: `version` is obsolete 
[+] Running 2/0
 ✔ Container object-detector-python-inference-server-1  Created                                                                                                                                        0.0s 
 ✔ Container object-detector-python-acap_dl-models-1    Created                                                                                                                                        0.0s 
Attaching to acap_dl-models-1, inference-server-1, object-detector-python-1
inference-server-1        | exec /opt/app/acap_runtime/acapruntime: exec format error
acap_dl-models-1          | exec /bin/sh: exec format error
Error response from daemon: failed to create task for container: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error mounting "/usr/lib/libvdostream.so.1" to rootfs at "/usr/lib/libvdostream.so.1": mount /usr/lib/libvdostream.so.1:/usr/lib/libvdostream.so.1 (via /proc/self/fd/6), flags: 0x5000: not a directory: unknown: Are you trying to mount a directory onto a file (or vice-versa)? Check if the specified host path exists and is the expected type

### Environment

- Axis device model: Axis P3255-LVE
- Axis device firmware version: 11.9.60
- Stack trace or logs: [e.g. Axis device system logs]
- OS and version: Ubuntu 20.04 LTS via VMware (same error when running Windows 11 natively)
- Version: See above

### Additional context

Add any other context about the problem here.

Corallo, 2024-04-10:
Hi,

First, I would recommend to downgrade the firmware to 11.8.
There is a [known issue](https://axiscommunications.github.io/acap-documentation/docs/release-notes/4.13.html) on 11.9 that will make your ACAP fail, we are working on it

But besides that, it seems you have another problem, because your acap is crashing here
`inference-server-1 | exec /opt/app/acap_runtime/acapruntime: exec format error`
exec format error usually occur when the architecture of the device doesn't match the one the ACAP was built for.

```
docker build --tag $APP_NAME --platform linux/arm/v7 --build-arg ARCH .
docker build --file Dockerfile.model --tag $MODEL_NAME --platform linux/arm/v7 --build-arg ARCH .
```
Here the platform flag is not required.

```
docker save $APP_NAME | docker load
docker save $MODEL_NAME | docker load
```
The purpose of this command is to save the docker container in your computer and send it to the camera, your docker load is missing the -H flag, double-check the readme of the example to see the correct usage

sreid-brock, 2024-04-12:
Thank you for your suggestions.

I'm using an Axis P3255-LVE which has linux/arm/v7 architecture. I'm also using Docker_Daemon_1_5_0_armv7hf_signed.eap on the device which I believe should match that architecture. Is there something else I'm missing? My Ubuntu machine is linux/amd64, which is why I added the --platform flag to the build command above, but I feel like I'm missing something here in terms of architecture/compatibility.

Unfortunately, the libvdostream error remains with the "P3255-LVE_11_8_61" firmware.

Please let me know if you have further advice.

Corallo, 2024-04-12:
armv7 and armv7hf are not the same architecture, and we define the architecture via the `--build-arg` flag, not the `--platform`.
Look again at the readme with the example build instruction:
https://github.com/AxisCommunications/acap-computer-vision-sdk-examples/tree/main/object-detector-python#how-to-run-the-code
Make sure to follow the build instructions as explained there, starting from the export of the variables.

sreid-brock, 2024-04-16:
The camera reports its architecture as armv7hf (through the basicdeviceinfo getAllProperties method of the API). So adding Docker_Daemon_1_5_0_armv7hf_signed.eap as an app on the device should work fine then, right? Are there other potential causes of the "exec format error" issue other than architecture, or something I'm still missing regarding the architecture?

I have been going through the instructions you linked with the same results. I must be missing something here.

Corallo, 2024-04-16:
Hi, the architecture of the docker acap is correct. 
Did you make sure to use the -H flag? for these three commands?
```
docker save $APP_NAME | docker load
docker save $MODEL_NAME | docker load
docker compose --env-file ./config/env.$ARCH.$CHIP up
```
Without the -H flag, you are running the example on your computer rather than on the device, that might be the problem

sreid-brock, 2024-04-18:
Thanks, I was able to make some progress by specifying the host (though I seemed to have a TLS issue so I've temporarily disabled it on the Docker daemon and switched to port 2375).

The save commands are working, but I'm getting a connection refused / authentication error with the docker compose command. Browsing to https://registry-1.docker.io/v2/ gives me the following error:

{"errors":[{"code":"UNAUTHORIZED","message":"authentication required","detail":null}]}

I'm able to successfully "docker login" on the command line so I'm not sure why there's an authentication error. Any thoughts?

docker save $APP_NAME | docker --host tcp://$DEVICE_IP:2375 load
docker save $MODEL_NAME | docker --host tcp://$DEVICE_IP:2375 load

docker --host tcp://$DEVICE_IP:2375 compose --env-file ./config/env.$ARCH.$CHIP up
			WARN[0000] /home/brock/Desktop/Axis/acap-computer-vision-sdk-examples/object-detector-python/docker-compose.yml: `version` is obsolete 
			[+] Running 1/0
			✘ inference-server Error Get "https://registry-1.docker.io/v2/": dial tcp: lookup registry-1.docker.io on [::1]:53: read udp [::1]:53711->[::1]:53: read: connection refus...                         0.0s 
			Error response from daemon: Get "https://registry-1.docker.io/v2/": dial tcp: lookup registry-1.docker.io on [::1]:53: read udp [::1]:53711->[::1]:53: read: connection refused

Corallo, 2024-04-19:
If the problem is TLS authentication, you can either disable TLS from the Docker ACAP settings,
Or (recommended for security) set up TLS certificates.
In the second case ou can follow the instructions here:
https://github.com/AxisCommunications/docker-acap?tab=readme-ov-file#use-tls

sreid-brock, 2024-04-19:
The error I posted above was with TLS already disabled in the Docker ACAP settings. I can set up TLS certificates, but isn't that unrelated to the "authentication required" error I'm getting at "https://registry-1.docker.io/v2/"?

Corallo, 2024-04-19:
Ah  I see so it could be that your camera doesn't have access to the internet directly?

One of the images needed, the acap-runtime is supposed to be pulled from the camera when you do the docker compose up command. If it can, you have to save it manually like you did for the other two containers.

try
```
docker pull axisecp/acap-runtime:1.3.1-aarch64-containerized
docker save axisecp/acap-runtime:1.3.1-aarch64-containerized | docker --host tcp://$DEVICE_IP:2375 load
```
and compose up again

sreid-brock, 2024-04-19:
Thanks for the suggestion, but I'm still getting the same "connection refused" error with the "unauthorized" and "authentication required" messages if I browse to the URL.

sreid-brock, 2024-04-19:
I've had someone else try and they cannot access https://registry-1.docker.io/v2/ either, so it's not just my account. Is this a private registry or something? I'm not sure why the camera is trying to access it or what I have to do to gain access.

sreid-brock, 2024-04-19:
Update: Your suggestion above was the right approach. I initially just didn't replace aarch64 with armv7hf in the command.

Thanks again for your help!

closed 2024-04-19 by sreid-brock