AxisCommunications/acap-computer-vision-sdk-examples #181: Issue obtaining images from camra source using a python script

AlejandroDiazD, 2024-04-25:
Hello, 

I'm trying to, using a python script, read an image from video streaming and later write it on my axis camera device. The objective is to process that image using a dnn similarly to the object-detector-python example, but my current issue is just about reading the image from video streaming.

I have followed the steps described in object-detector-python example, but for the following image (screenshot from the web interface of the camera):

[image]

I'm getting the following image:

[image]

Next, I provide my specific case characteristics:
* Camera model: AXIS P3268-LVE Dome Camera
* Camera firmware: 10.12.165
* Camera resolution: 1280x960 (4:3). This is the resolution used for the provided images, but I have already tried the same case for different resolutions (with similar results).

I also provide some reproducible code:
> import cv2
> from vdo_proto_utils import VideoCaptureClient
> stream_width, stream_height, stream_framerate = (1280, 960, 10)
> grpc_socket = 'unix:///tmp/acap-runtime.sock'
> capture_client = VideoCaptureClient(socket=grpc_socket, stream_width=stream_width, stream_height=stream_height, stream_framerate=stream_framerate)
> frame = capture_client.get_frame()
> cv2.imwrite('/app/output_image.jpg', frame)

I'm running that code inside a docker container based on a docker image built using the Dockerfile provided in the object-detector-python example.

Thank you for your time.

AlejandroDiazD, 2024-04-25:
Update: I almost solved the issue (@Corallo ).

The problem was related to the image rotation of the camera. To solve it I changed the rotation from the web interface in /video/installation.

Now the only problem is the color representation, we can see how the blue parts of the image are seen as brown.

The original image:

[image]

The obtained image:

[image]

Do you know how can I solve it?

Thanks!

Corallo, 2024-04-25:
Hi @AlejandroDiazD 

Good that you solved the issue by yourself,
regarding the color problem, I think it is just that cv2 expects the color format in bgr when you save the image, while your frame is in rgb.
try
`bgr_img = cv2.cvtColor(frame, cv2.COLOR_RGB2BGR)`

AlejandroDiazD, 2024-04-26:
It worked! :)

Thank you @Corallo, feel free to mark the issue as solved when you consider.

We keep in contact.

closed 2024-11-04 by pataxis