AxisCommunications/acap-computer-vision-sdk-examples #190: Spike in inference time from the inference server

akash-syook, 2024-08-21:
> **Please do not disclose security vulnerabilities as issues. See our [security policy](../../SECURITY.md) for responsible disclosures.**

### I have trained yolov5m model and sucessfully deployed on Axis "AXIS M4317-PLR Panoramic Camera". 
The avg inference time is 70-80 ms but sometimes there is a spike in inference time  from inference server.
I am using below repo for my custom trained yolov5 model
https://github.com/AxisCommunications/axis-model-zoo/blob/main/docs/yolov5-on-artpec8.md
For deployment I have used
https://github.com/AxisCommunications/acap-computer-vision-sdk-examples/tree/main/object-detector-python

Added sample screenshots. 
Please let me know what could be wrong.
[image]
[image]

Explain the behavior you would expect and the actual behavior.

### To reproduce

Please provide as much context as possible and describe the *reproduction steps* that someone else can follow to recreate the issue on their own.

A team member will try to reproduce the issue with your provided steps. If there are no reproduction steps or no obvious way to reproduce the issue, the team will ask you for those steps. Bugs without steps will not be addressed until they can be reproduced.

Steps to reproduce the behavior:

1. Set up '...'
2. Do this '...'
3. See error

### Screenshots

If applicable, add screenshots to help explain your problem.

### Environment

- Axis device model: [e.g. Q1615 Mk III]
- Axis device firmware version: [e.g. 10.5]
- Stack trace or logs: [e.g. Axis device system logs]
- OS and version: [e.g. macOS v12.2.1, Ubuntu 20.04 LTS, Windows build 21996.1]
- Version: [version of the application, SDK, runtime environment, package manager, interpreter, compiler, depending on what seems relevant]

### Additional context

Add any other context about the problem here.

pataxis, 2024-09-19:
Hi @akash-syook , what version of AXIS OS and ACAP Runtime are you using? It looks like it's the template values for the Environment in your question.

akash-syook, 2024-09-27:
Axis os: 11.11.73
ACAP Runtime: 1.3.1
Let me know if you need any inputs.

Stiv-work, 2024-11-01:
Hi @akash-syook 
We could unfortunately not replicate the issue.

closed 2024-11-01 by Stiv-work